irungentoo/Xiaomi_Yi_4k_Camera #2: thanks + question

monouser7dig, 2017-03-30:
thank you very much for sharing this!

one question though
can this break the camera in a way that can not be restored by re-applying the firmware update or in some other way?
(probably yes, but, is it likely to happen?)

thanks!

irungentoo, 2017-03-31:
The scripts just patch out values in ram so the chances of them bricking the camera are pretty much impossible if you run them on the right firmware version.

Removing the script from the sd card is the only thing you need to do to remove the mod.

monouser7dig, 2017-03-31:
ok thanks

actually the remaining recording time does not change with/ without the file.
tried 2,7k 60p with/ without and the same time remaining 
when I record a test sequence it comes out at about 60MBit/s In both cases 

I set the quality to high.

it's the Z16V13L
1.4.0 en-us build 2395r

irungentoo, 2017-03-31:
Be sure to "save page as" instead of just copy pasting and creating a new file if you are on windows like the readme says.

The file needs to have unix line endings and an empty line at the end for the camera to run it.

monouser7dig, 2017-03-31:
I downloaded the RAW, changed from txt to ash but although my computer showed just then .ash part it was actually .ash.txt so I had to reserve that and now it's up and running

sorry 😆 
thanks again, I'll try some sort of quality comparison and if I find something, report back!

Vitold69, 2017-04-06:
monouser7dig  just right click on the autoexec.ash link on the page and choose save as... option. It will save autoexec.ash file without any changes. Then copy it to the root of your sd card. Hope this helps.

irungentoo, great work! Please do not stop!

monouser7dig, 2017-04-06:
> it's up and running

I actually shot some test scenes and could not see a difference 
a little bit of difference in the noise of dark tones at 600% zoomed in but....nothing serious 

Tried 4k 2,6k in a 4k and 1080p timeline both at 60p using normal pictures profile with low sharpening in camera and post process sharpening and small color adjustment to stretch out the histogram.