irungentoo/Xiaomi_Yi_4k_Camera #3: PAL resolution

zgmladen, 2017-04-05:
Please add addresses for PAL resolution

irungentoo, 2017-04-08:
I checked quickly and for the bitrate mod the addresses are the same for PAL. 

Ex: if you want to change the bitrate for the 2.7K mode at 50p for pal, use the address for the 2.7k mode at 60p.

TijuanaKez, 2017-05-20:
For me, the bitrate mods only seem to work in 4K / 4K Ultra in PAL mode. 2.7K/25p is still 45mbps, 1080p ~30mbps.

irungentoo, 2017-05-22:
@TijuanaKez 

If you want to mod the bitrate of the 2.7k/25p mode you need to mod the bitrate of it's equivalent in NTSC which is the 2.7k/30p mode. Same thing for 1080p.

TijuanaKez, 2017-05-22:
> If you want to mod the bitrate of the 2.7k/25p mode you need to mod the bitrate of it's equivalent in NTSC which is the 2.7k/30p mode. Same thing for 1080p. <

That's what I gathered from this thread, just it doesn't to work. I'm just using the premade bitrate mod script in the bitrate_mod folder. The 4K stuff comes out at 100mbps, but the 2.7k/25p at least is still 45mbps.