irungentoo/Xiaomi_Yi_4k_Camera #4: SD Card Fix as a autoexec.ash file?

Vitold69, 2017-04-06:
Is it possible to make the SD Card fix in the same way as bitrate mode i.e. in autoexec.ash?
Thanks! Keep the good work!

irungentoo, 2017-04-06:
It would not work because if the camera can't read the sd card it can't read the autoexec.ash

Unless you want to put the autoexec.ash file on a sd card that works by default and swap to the other card every time you open the camera.

Vitold69, 2017-04-06:
irungentoo thanks for the reply!
One more question. In your bitrate table I've found something interesting. There are some resolutions which do not present in the official 1.4.0 firmware. They are like 0xA061B08E 2688x2688 30p and very interesting  0xA061B20E 2560x1440 30p hdr. HDR!? 
Could you please provide a resolution substitute table as you did for 4000x2700 mod, i.e. what address to record what to get the 2560x1440 30p hdr for example.
Thanks a lot!

irungentoo, 2017-04-07:
For the 4000x2700 mod I just changed the resolution in the resolution table of the 4k ultra mode to 4000x2700.

I have not yet looked into how to use modes that are in the firmware but not accessible in the GUI.

closed 2017-04-07 by Vitold69