irungentoo/Xiaomi_Yi_4k_Camera #7: thx! +live stream & ultra 4k

andyspeaking, 2017-04-10:
Thanks a.... LOT for your work and for sharing this!
Unfortunately in PAL the ultra 4k remains on 3840x2160. Could you tell me the proper "writel" line please? I know that it is the same 24p but it would avoid me from swapping pal->ntsc->pal when using other resolutions.

I haven't found a resolution-fps table to work in conjunction with your bitrate table to do it myself..sorry.

As for the stream...I would like to receive a rtsp one, like I did with prevoius yi, to stream on VLC. Would it be possible by using the Live+qr code feature someway? I really would like to share the stream on a wifi network instead of using an ad-hoc YDXxxxxx classic app connection.

Thanks again for your great work!

irungentoo, 2017-04-10:
This should make the 4k ultra mode work on PAL: https://gist.github.com/irungentoo/a87c0df27c13d6603aca6c864c06b871

I know that if you use the yi app the video is transferred to the app using an rtsp stream but I have not yet looked at how to watch it with something else.

If you use the live mode you can make it stream to any rtmp server (I use: https://github.com/arut/nginx-rtmp-module and it works well) and then watch the stream by pointing your media player at that server.

andyspeaking, 2017-04-10:
thanks! I'll take a look to nginx... do you think I could set a windows based server with it? I'm trying to find a portable solution to have the 4k rtmp feature. Thanks again!