irungentoo/Xiaomi_Yi_4k_Camera #9: Modify sdcard fix firmware

tictacmoe, 2017-04-11:
Hello and big thanks for the great work!

I am trying to use a Toshiba EXCERIA M302-EA 128GB with the Yi 4k and suspect the same problem as you had with the Lexar 633x, wrong access frequency.

I would therefore patch mtdblock1 at the corresponding location.
As directly writing to /dev/mtdblock1 over telnet is not working (Operation not permitted) and not permanent, I am interested in how you managed to modify the firmware image.
Did you unpack & repack the firmware? Which tools were you using for achieving this result?

Thanks!

irungentoo, 2017-04-12:
The firmware isn't compressed or anything so if you can figure out where the part you want to modify is you can just modify it with a hex editor.

The 4 last bytes of the firmware file are a crc32 checksum that the camera checks before flashing the firmware. There are also crc32 checksums for each section of the firmware.

The camera prints in the log the expected checksum and the checksum that it got so you just need to try to flash your modified firmware and then find replace the expected checksum with the checksum it got (Just remember that the numbers in the firmware are little endian so they are "upside down").

tictacmoe, 2017-04-12:
Thanks for the great explanation.
I could successfully modify and update the firmware. This got my micro sd card working. The fix will probably also work for other Toshiba cards. I did not have to change any section crc32 checksum btw.

If anyone else wants to use Toshiba Exceria M302-EA 128GB cards, you can download my patched firmware from the link below. The patch modifies the access frequency for Toshiba cards to 100MHz.
https://mega.nz/#!SYEwEaAD!EEmxPUT7mzZGub8Yz3FZPBp2sm8CmpAwtbPtL9GZmVI

This is the en_US version 1.40 firmware for serial number Z16V13L.
Only use it, if you know what you are doing! I am not responsible for any damage to your device.

closed 2017-04-12 by tictacmoe