irungentoo/Xiaomi_Yi_4k_Camera #10: Firmware 1.4.14 released, bitrate hack no longer works

hirsti, 2017-04-12:
Hi,

Thanks for all your hard work.
Yi have released a new firmware, version 1.4.14
The bitrate hack no longer works.
could you have a look please..

I would love to know how you determined the addresses in the first place :smile:

irungentoo, 2017-04-12:
It's updated to latest, I did it quickly so let me know if something doesn't work.

I found them using a hex editor and by checking how bitrate mods for other ambarella cameras worked. Updating it is pretty simple once you have already found them in an older firmware.

digitalhecht, 2017-04-20:
Thanks for doing this nice work. I am having a hard time getting the bitrate mod to work. I am a bit confused: The addresses listed in the .ash file don't match what's listed in the bitrate table (.txt) file. Can you drop a hint? :)
Thanks!

irungentoo, 2017-04-21:
Yes, they match. I just didn't put the addresses for the medium and low settings in the bitrate table because it's just the high setting with 0x10 and 0x20 added to it.