irungentoo/Xiaomi_Yi_4k_Camera #11: Having trouble downloading the file under windows.

mafutha, 2017-04-15:
Could it be possible to rar/zip the file so all those line ending would not Vanish? I tied to use notepad++ but couldn't figure out the format to asve it as.

tictacmoe, 2017-04-15:
I don't know exactly which file you mean, but you can download the entire repository through
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/archive/master.zip
You can find this link under the green "Clone or download" button on the main repository page.

If you view a file in RAW and save it, this will also preserve the line endings. No further need for converting.

In notepad++ you can convert line endings under Edit --> EOL Conversion -> Unix (LF)
If this menu item is grayed out, than the file already has the correct line endings.

mafutha, 2017-04-15:
Thanks for the response. I was only interested in autoexec.ash file for the bitrate changes but the whole archive is just as good. I appreciate thew info using notepad++

Thanks again...Terry