irungentoo/Xiaomi_Yi_4k_Camera #12: The right SD Card?

digitalhecht, 2017-04-21:
Just an observation: I was able to get the bitrate hack to work so far on only one card out of three. My SanDisk 64GB SDXC I Class 10 (U1) worked (for 30 seconds before getting the "Slow Card" error).
My PNY 64GB SDXC I Class 10 (U3) and Lexar 16GB SDHC I Class 10 (U1) were both no go's... (I was a bit surprised that the SanDisk U1 worked over the PNY U3, albeit 30 seconds. But I don't think the PNY's are generally well-reviewed, despite that higher speed designation.)
The way I tested was checking the bitrate in GSpot afterwards. 
I plan to get a higher end SanDisk U3 card later today. (Anything to hold off that Yi 4K+ purchase. I have more need of higher bitrate 2.7K/60 fps for the medium FOV anyway.)

irungentoo, 2017-04-21:
Try lowering the bitrates.

digitalhecht, 2017-04-21:
Ah! That did the trick! I lowered the bitrate to 90 Mbps for 2.7K 60 fps and it worked! Thank you. My ideal situation is to find a card fast enough to support 120 Mbps at the same resolution in order to get 60 Mbps at half speed. Thanks again. This will definitely tide me over!

closed 2017-04-21 by digitalhecht