irungentoo/Xiaomi_Yi_4k_Camera #13: Firmware 1.4.23 released

monjahrenel, 2017-04-27:
Hi. was wondering if you will update your tweaks / hacks to support the newly released firmware? Thanks!

irungentoo, 2017-04-27:
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/blob/1.4.23/my_autoexec.ash/autoexec.ash

This is for firmware 1.4.23, I'll update the rest of the files when I have time.

monjahrenel, 2017-04-27:
wow that was fast! will wait for your 1.4.23 update on sd card issue irungentoo. thank you so much!

zgmladen, 2017-04-27:
Firmware 1.4.23 update is for multilingual. Will hack work with that firmware?

irungentoo, 2017-04-27:
@monjahrenel they actually fixed the sd card bug in the 1.4.23 firmware.

irungentoo, 2017-04-27:
Everything is updated to 1.4.23 now.

zgmladen, 2017-04-28:
2.7k 60p (in PAL) bitrate is variable

sylido, 2017-04-28:
Shouldn't the high quality bitrate for 4k 30p in the autoexec.ash file be 0x42C0 it's currently 0x42CA which is not in the custom bitrate file ?

irungentoo, 2017-04-29:
@sylido 0x42C0 is just a number in hexadecimal, you can use numbers between the values I give in the bitrate file like what I did there. I set the bitrate to the maximum that my sd card supports which was slightly above the 0x42C0 value.

leftbrasil, 2017-04-30:
@irungentoo For me, your autoexec didn`t worked. What i did:
1-Formated SD Card thru the camera menu;
2-Right click and save asas at the file located at https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/blob/master/my_autoexec.ash

Right click on video files shows 60Mbps when recorded with 2.7k and High quality options. Also, the recording time available didn`t change.
In the previous firmware, i was using you mod perfectly.
Thanks for your help!

Vitold69, 2017-04-30:
You need to rename the my_autoexec.ash to autoexec.ash

leftbrasil, 2017-04-30:
@Vitold69 It`s already autoexec.ash. The link that i shown are related to a page, not a file. The file im my SDCard root is autoexec.ash.

MidoriYakumo, 2017-05-23:
1.5.1 released :( what the hell is sd card optimization

Vitold69, 2017-05-24:
Yeah... Another shit from YI. It leaves an empty "test" directory after itself.  Probably tests read/write speed of an SD card.

MidoriYakumo, 2017-05-24:
@Vitold69 Maybe it detects best writing block size by such a test?

irungentoo, 2017-05-24:
What's interesting is that in the 1.5.1 firmware they seem to have added a bunch of 24p modes but they are not exposed yet in the UI.

irungentoo, 2017-05-25:
It's updated to 1.5.1

MidoriYakumo, 2017-06-25:
1.6.1 RAW is good, XiaoMi optimized some buffer policy making it possible up to 100Mbps for 4k by default
150Mbps bitrate get longer working time on my sdcard :)
however 720p 240 stuck even in 100Mbps wtf??? :(

My autoexec for 1.6.1

# Yi 4k bitrate mod, 4k ultra 4000x2560 and 1080p ultra 90p 2000x1504 originally by irungentoo
# For firmware 1.6.1

# NamingMod
# YDXJ -> YN00
writel 0xA05000F4 0x30304e59
# YDXJ -> YT00
writel 0xA0500104 0x30305459
# YPXJ -> YP00
writel 0xA0500110 0x30305059
# YSXJ -> YS00
writel 0xA0500120 0x30305359

# start from DCIM@0xA0646BC8
# 0x42F0 120Mbps high quality, 0x42B4 90Mbps medium, 0x4278 60Mbps low
# 4k 30p
writew 0xA0646C9E 0x42F0
writew 0xA0646CAE 0x42B4
writew 0xA0646CBE 0x4278
# 4k
writew 0xA0646CCE 0x42F0
writew 0xA0646CDE 0x42B4
writew 0xA0646CEE 0x4278
# 4k ultra 24p
writew 0xA0646CFE 0x42F0
writew 0xA0646D0E 0x42B4
writew 0xA0646D1E 0x4278
# 2.7k 60p
writew 0xA0646D8E 0x42F0
writew 0xA0646D9E 0x42B4
writew 0xA0646DAE 0x4278
# 2.7k
writew 0xA0646DEE 0x42F0
writew 0xA0646DFE 0x42B4
writew 0xA0646E0E 0x4278
# 1080 120p
writew 0xA064798E 0x42F0
writew 0xA064799E 0x42B4
writew 0xA06479AE 0x4278
# 720 240p
writew 0xA0647A1E 0x42F0
writew 0xA0647A2E 0x42B4
writew 0xA0647A3E 0x4278
# 1080 ultra 90p
writew 0xA0647CBE 0x42F0
writew 0xA0647CDE 0x42B4
writew 0xA0647CEE 0x4278

# Set NTSC 4k ultra to 0FA0 4000 x 0A00 2560
writel 0xA064DACC 0x0A000FA0
# Set NTSC 1080p ultra 90p to 07D0 2000 x 05E0 1504 and 4:3
writel 0xA064E2AC 0x05E007D0
writew 0xA064E2B0 0x0403

vipya, 2019-04-15:
I am using yi 4k with 1.10.9 firmware. Please tell me whole process of updating bitrate.