irungentoo/Xiaomi_Yi_4k_Camera #15: Video mod is unstable

MidoriYakumo, 2017-05-26:
I currently(1.4.23) changed bitrate to 90M/75M/60M, resolution to 4000x2560, 2000x1500 based on my_autoexec.ash(1080p ultra 90p not touched from irungentto version)
It works pretty fine usually. However when switched to 1080p ultra 90p, LCD display sometimes become 4:3 ratio video preview, in this case I start recording and the screen goes black and stucked.
Any idea?

MidoriYakumo, 2017-05-26:
strings from firmware and some help output from ambashell
has some difference with gopro and xiaomi yi, might be help

[help.zip](https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/files/1030637/help.zip)

irungentoo, 2017-05-26:
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/commit/ea6eb81954b55a8abc2a77dcbd743caa867421a2

Maybe this will fix it, it makes it so you see a 4:3 preview instead of a 16:9 one when you record in those modes.

MidoriYakumo, 2017-05-30:
1.5.1 with fix still crashes in 4:3 preview, don't know why :(