irungentoo/Xiaomi_Yi_4k_Camera #21: Resolution 3840x2880

Aztar, 2017-06-26:
Hi,
can you please add address and code for 3840x2880 (proper 4:3) resolution for this and future firmwares so the 4k ultra mode can be modified to this resolution? 

so this:
#set 4k ultra to 4000x2700
#NTSC
writel 0xA064DACC 0x0A8C0FA0
writew 0xA064DAD0 0x0403
#PAL
writel 0xA065444C 0x0A8C0FA0
writew 0xA0654450 0x0403

how to make to be this:
#set 4k ultra to 3840x2880
#NTSC
writel ?
writew ?
#PAL
writel ?
writew ?

or can you explain how to do it myself?

Thank you.

irungentoo, 2017-06-26:
0x0A8C0FA0 -> 0x0A8C and 0x0FA0.
0x0A8C = 2700 and 0x0FA0 = 4000

So if you want 3840x2880
2880 = 0x0B40
3840 = 0x0F00

So you would put: 0x0B400F00 instead of 0x0A8C0FA0 to get 3840x2880

Aztar, 2017-06-26:
Great. Thank you.
Also, thank you for all the work you are doing here and sharing it with us.