irungentoo/Xiaomi_Yi_4k_Camera #25: I got one 4k+, cannot figure out mtdblock1 memory offset

MidoriYakumo, 2017-08-01:
The 32M mtdblock1 is almost same with Yi4k, how ever I don't know where to get the memory offset.
as H2 is 64-bit soc, 0xa0000000 not work for it, this is dmesg head, any idea?
Left: A9ES5, right: H2
[image]

MidoriYakumo, 2017-08-01:
There is API to change bitrate and codec(to h265), however stuck the camera while recording:

> t app test chg_bitrate c [bitrate]

> t app test chg_bitrate [avg_bitrate][min_ratio][max_ratio]

> Usage: t app test record <option>
    Test option list:
	 show
	 dual  [0 | 1] 
	 duals [0 | 1] 
	 enctype [streamId][encodeType] 
	     streamId:   0 -> primary, 1: secondary 
	     encodeType: 0 -> H264,    1: H265

irungentoo, 2017-08-01:
dmesg won't show you anything relevant because the bitrate stuff is all done in the RTOS, not on the linux.

Try using the RTOS ps command if it exists on the yi4k+.

MidoriYakumo, 2017-08-03:
ps outputs:

```
 ID  PRI  PREEMPT  SCHED            STAT  NEXT_ID  STKSIZE  STKPEAK  STKUSED  STKBASE     STKEND      STKPTR      SMP  NAME

---------------------------------------------------------------------------------------------------------------------------------------

  0    0        0   1860       SUSPENDED        0     1020      656      480  0x02d9941c  0x02d99817  0x02d99638  0x3  System Timer Thread

  1  254      254      7      EVENT_FLAG        8     8188      708      596  0x00eb1300  0x00eb32fb  0x00eb30a8  0x3  RootTask

  2   32       32    788      EVENT_FLAG        2     3068     2012      644  0x0118d760  0x0118e35b  0x0118e0d8  0x1  AmbaSD_CardMonitorTask

  3   70       70    549       COMPLETED        3    65532    12916      396  0x01a6c048  0x01a7c043  0x01a7beb8  0x1  App_boot_Manager

  4  202      202   1013           SLEEP        4    40956      812      636  0x00ea7300  0x00eb12fb  0x00eb1080  0x2  AmbaPrintTask

  5   25       25      1      QUEUE_SUSP        5    32764      584      660  0x531e1800  0x531e97fb  0x531e9568  0x1  MW_FifoTask

  6   73       73      5      EVENT_FLAG        6    16380      692      580  0x53cabb80  0x53cafb7b  0x53caf938  0x1  MW_CFS_Scheduler

  7  160      160      8      EVENT_FLAG        7    16380     7316      588  0x53cb3600  0x53cb75fb  0x53cb73b0  0x1  MW_DCF

  8  119      119      4      EVENT_FLAG        1    32764      800      612  0x00c71400  0x00c793fb  0x00c79198  0x1  Monitor_SdTask

  9  175      175      3           SLEEP        9     4092      644      532  0x00c79920  0x00c7a91b  0x00c7a708  0x1  AppLib Monitor Storage

 10  150      150      3      EVENT_FLAG       10     4092      948      652  0x02e24150  0x02e2514b  0x02e24ec0  0x3  BQ27421_Manager

 11  108      108    676  SEMAPHORE_SUSP       11    65532    41336      772  0x01de52e8  0x01df52e3  0x01df4fe0  0x1  AppLib_Host_Control_Manager

 12  250      250      2  SEMAPHORE_SUSP       12     4092      732      684  0x00ca42f0  0x00ca52eb  0x00ca5040  0x2  AmbaShell_CliTask

 13  249      249    134           READY       13    32764    13756     8884  0x544f66c0  0x544fe6bb  0x544fc408  0x2  AmbaShell_ExecTask

 14  166      166      6           SLEEP       14    65532      668      556  0x5462a840  0x5463a83b  0x5463a610  0x2  QRManager

 15  166      166     24           SLEEP       15    65532      848      532  0x5476a5c0  0x5477a5bb  0x5477a3a8  0x2  RTMPServer

 16  170      170      1      EVENT_FLAG       16    32764      288      580  0x01cd9e98  0x01ce1e93  0x01ce1c50  0x2  VoiceCmd_Task

 17   20       20     70      EVENT_FLAG       17    16380     4132      608  0x54b9ba2c  0x54b9fa27  0x54b9f7c8  0x1  AmbaImgAeCtrlTask#0

 18   30       30     66      EVENT_FLAG       18     8188     1416      608  0x54b9ffa4  0x54ba1f9f  0x54ba1d40  0x1  AmbaImgAwbCtrlTask#0

 19   50       50     96      EVENT_FLAG       19    32764     4208      608  0x54ba251c  0x54baa517  0x54baa2b8  0x1  AmbaImgAdjCtrlTask#0

 20   45       45      2      EVENT_FLAG       20     8188      488      584  0x54baaa9c  0x54baca97  0x54bac850  0x1  AmbaImgObCtrlTask#0

 21   59       59    371      EVENT_FLAG       21     8188      844      732  0x551a2478  0x551a4473  0x551a4198  0x2  Audio Output Task

 22  105      105    376  SEMAPHORE_SUSP       22     8188      904      720  0x551ad534  0x551af52f  0x551af260  0x2  Audio Decoder Task

 23  131      131     17      EVENT_FLAG       23    16380     1188      940  0x552b8e80  0x552bce7b  0x552bcad0  0x2  _AppLibAudio_BeepTask

 24   10       10      1  SEMAPHORE_SUSP       24     8188      328      636  0x010df800  0x010e17fb  0x010e1580  0x2  AmbaImg_EisProcTask

 25   18       18      1  SEMAPHORE_SUSP       25    65532      240      532  0x010e1800  0x010f17fb  0x010f15e8  0x2  AmbaImg_EisProcAlgo2Task

 26  255      255     21  SEMAPHORE_SUSP       26     4092      636      524  0x55501cc0  0x55502cbb  0x55502ab0  0x1  AmbaUSBDeviceVbusCheckTask

 27   20       20    100      EVENT_FLAG       27     4092      848      628  0x55502d00  0x55503cfb  0x55503a88  0x1  MW_VinHandler#0

 28   54       54    292      QUEUE_SUSP       28    65532    33244      740  0x55506d80  0x55516d7b  0x55516a98  0x1  MW_VideoEncode

 29   62       62      1      QUEUE_SUSP       29    65532      936      756  0x5c406040  0x5c41603b  0x5c415d48  0x1  MW_StillEncode

 30   60       60      1      QUEUE_SUSP       30    32764      840      748  0x5c45bc40  0x5c463c3b  0x5c463950  0x1  MW_StillRawCap

 31   64       64      1      QUEUE_SUSP       31   131068      992      780  0x5c465840  0x5c48583b  0x5c485530  0x1  MW_StillRawEncode
[... 1662 lines cut ...]
 51        130       0.0%  LINK_RPMSG_RVQ
 52        441       0.0%  LINK_RPMSG_SVQ
 53        308       0.0%  HookDefaultUserRPCTask
 54          5       0.0%  LINK_RPC_SVC_THREAD 0
 55          4       0.0%  LINK_RPC_SVC_THREAD 1
 56          6       0.0%  LINK_VFS_OPS
 57          3       0.0%  LINK_VFS_SG
 58          4       0.0%  LINK_HIBER
 59          8       0.0%  LINK_RPC_SVC_EVENTNOTIFIER
----------------------------------------------------
 Core-0:
----------------------------------------------------
         56945       1.2%  Interrupt Service Routine
       2410492      50.3%  Task
       2293459      47.9%  Idle
         29299       0.6%  Overhead
----------------------------------------------------
 Core-1:
----------------------------------------------------
          3728       0.1%  Interrupt Service Routine
        104221       2.2%  Task
       4665024      97.4%  Idle
         17222       0.4%  Overhead
----------------------------------------------------
```

irungentoo, 2017-08-03:
From that PS output (STKBASE) I see you have 2 memory offsets, 0x00000000 and 0x50000000.

If I compare with the output I had from my yi4k I think that you should try dumping the memory at: 0x00000000 and see if the bitrate table is somewhere in there.

To find it just look for 00 00 70 42 and you should be able to find it easily if it's there.

MidoriYakumo, 2017-08-04:
nope, I have dumped all available(skip if fails) 16M blocks from 0x0 to 0xffffffff (32-bit addresses), found no desire content:

```shell
PS A:\AmbaDump> grep 240fps .\mtdblock\mtdblock1
Binary file .\mtdblock\mtdblock1 matches
PS A:\AmbaDump> grep 240fps .\dump\*
PS A:\AmbaDump> grep superview .\mtdblock\mtdblock1
Binary file .\mtdblock\mtdblock1 matches
PS A:\AmbaDump> grep superview .\dump\*
PS A:\AmbaDump> dir .\dump\

    Directory: A:\AmbaDump\dump

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----         8/3/2017   2:27 PM              0 dump0x0000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x10000000_16M.bin
-a----         8/3/2017   2:27 PM       16777216 dump0x1000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x11000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x12000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x13000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x14000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x15000000_16M.bin
-a----         8/3/2017   2:31 PM       16777216 dump0x16000000_16M.bin
......
```

0x0, 0x2000000, 0x80000000~0xffffffff cannot be read out

irungentoo, 2017-08-04:
If you try to dump: 0x01000000 does it work?

MidoriYakumo, 2017-08-04:
-a----         8/3/2017   2:27 PM       16777216 dump0x1000000_16M.bin

there it is

only 0x00000000, 0x02000000  fail

irungentoo, 2017-08-04:
if you upload the dumps somewhere I can take a quick look.

MidoriYakumo, 2017-08-04:
https://drive.google.com/file/d/0B8IzIzeTTCbITEM5dzd2VFRsQmc/view?usp=sharing

irungentoo, 2017-08-04:
If you try dumping: 0x00100000, does it work?

Theangelofspace15, 2017-08-09:
Hi. What language is that? Os that assambly? I want to learn how to mod my yi amd I want to know what to learn.

MidoriYakumo, 2017-08-23:
Awesome! C:\DCIM is at 0xA22934 @irungentoo 
[image]

closed 2017-08-24 by MidoriYakumo