irungentoo/Xiaomi_Yi_4k_Camera #29: VR firmware change resolution

nhamg, 2017-09-04:
Great to see the new VR stuff.

Sadly I have no success changing the video resolution - is it possible inside the VR firmware?

My current attempt:
writel 0xA05FAA6A 0xA05FAACA

(1.0.179_en-us_firmware_first)

Thank you.

irungentoo, 2017-09-05:
The 4k ultra res on that firmware is 30fps and if I'm reading this correctly it reads a 4096x2160 image from the sensor and then downscales it to 3840x2160.

On the normal firmwares it reads the full 4000x3000 image which is why the ultra res is only 24fps and why the resolution can be extended a lot.

The address for the 4k ultra res is: 0xA0600700 but the max res it can have is: 0x08701000 (4096x2160) 
If it doesn't work: 0x08700FA0 (4000x2160) might.

the command would be:
writel 0xA0600700 0x08701000

nhamg, 2017-09-08:
I see, interesting info about internal resolution and actual output res! Sadly, I had no success with either suggested option for the autoexec. But then I can't seem to get any autoexec stuff to work on the VR firmware (yet having UNIX line endings and last line free), I might be doing something wrong...
Thanks a lot!

irungentoo, 2017-09-09:
I did a bit of testing and for the autoexec.ash to work you need to have a valid .script file in the root of your sd card.

For example save_log_enable.script which enables logs or console_enable.script which enables telnet make it work.