irungentoo/Xiaomi_Yi_4k_Camera #33: RTSP Stream resolution

Nukersson, 2017-10-23:
Hello,

You you probably now how RTSP-Stream resolution can be changed? (RTSP works on rtsp:\\192.168.42.1\live). I can only reach 640x480 or 848x480. 
It would be nice to get 1920x1080 or 1280x720.

Very good work although.

Thanks,
Alex

MidoriYakumo, 2017-10-27:
do you use official app qr generator?
I made 1080p stream with it

l30ns, 2017-11-07:
Hi,
Is there way to use custom QR code to stream hi-def to some switcher software like Wirecast or such? They all take RTMP streams, but i cant seem to figure out how to configure the Yi 4K part.

sukkamehu, 2017-11-08:
https://www.dropbox.com/s/e1w0y7rtsw21b28/Z18V13L_1.3.18_build-20171013171743_git-b1438f82_r981.bin?dl=0

With newest beta firmware (link above) it's possible to rtsp stream as a wifi client on YI4k+. I used barcode below:
{"res":"1080p","sign":"0","url":"127.0.0.1:554/live","ak":"0","ssid":"WiFi","pwd":"12345678","rate":"5","dur":"0"}

Just don't press reconnect button when camera say "Live connection failed". It's still working behind. Also if you press back two times connection stops.

Feel free to try my qr generator to try it
http://80.211.209.99/yi/

Nukersson, 2017-11-18:
Thank you sukkamehu,

The firmware works fine and RTSP starts as you described. But it is still 848 x 480 - so it is not 1080p. I have used your barcode creator - and resolution can not be changed.

Cheers,
Alex

sukkamehu, 2017-11-18:
Thanks Nukersson! I also noticed that same issue, I'll try to 'reverse engineer' it probably next week so if I'll find some camera sent json output, I'll change it to that command.

I think it could be possible to get camera stream better rtsp stream but first I need to know how does that streaming to fb and youtube works. I'll inform here once I've tested if it's possible or not.

If anyone know some action camera which can stream +720p as wifi client I'd like to know.

Juuso

Nukersson, 2017-11-18:
Thank you Juuso,

Be careful - streaming to Facebook or YouTube works as RTMP stream. Therefore if you will convert RTMP to RTSp you'll get delay. And this delay could be up to 5 seconds.
I need RTSP with almost less than 150 ms. This is the challenge.
I have written with an engineer from Yi - he means it is possible to unlock 1080p RTSP stream and they could do it in Ambarella SDK, but won't. WTF.

Hope you'll find it.

sukkamehu, 2017-11-18:
Thank you really much for your answer!

In my project as well it would be also too much if delay is more than 200 ms, I didn't think enough how much converting takes time. Could help if I would have full list of available commands or any documentary somewhere.

Really, what the f?! Why couldn't they answer in yi how to open 1080p for streaming if it's somehow possible.. If you can provide me email of that engineer I could also try to ask. For me only reason to got that camera was possibility to stream as wifi client.

Nukersson, 2017-11-18:
Hey Juuso,

I think it is the restriction of Ambarella - the chip manufacturer for YI 4k+. And you'll not be able to buy the developer board with an SDK since they need you buy more then 15k chips / year. 

Just write to YI's support and state your problem.

Hope you'll get it solved,
Best Regals,
Alex

irungentoo, 2017-11-19:
I think the rtsp resolution is probably hardcoded somewhere in the firmware RTOS code and the stream is probably the same as the one that gets put in the .SEC files.

sukkamehu, 2017-11-21:
With newest firmware better resolutions are working!

Download link can be found here:
https://www.reddit.com/r/yi4kplus/comments/7eg7nv/yi_4k_november_9_2017_beta_firmware_version_1322/

Nukersson, 2018-01-31:
I've tried it. It does not work too.

Triuman, 2018-02-01:
What is the delay do you get at 720p live stream? I also need a latency lower than 150 ms.

Triuman, 2018-02-12:
@Nukersson Did you get any result with low latency streaming? When I stream using QR code, I see 1-2 seconds delay via VLC Player. I desperately need latency below 200 ms.

Triuman, 2018-02-12:
I finally figured it out. Please check the issue below.

[https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/issues/56](url)

alex-rptrv, 2018-02-26:
Can’t see your link

Triuman, 2018-02-26:
That is weird. The issue number is 56. You can check that out from the list.

SarHawk, 2018-03-14:
Any fixes on the Resolution Problem found?

Nukersson, 2018-03-14:
@Triuman Hey. Sorry for very delayed answer. Since the hardware of Yi 4k+ does not deliver expected results and there is no support from chipmaker of cameras processor, I've switched to other hardware. Therefore it is my last time I am writing here.

My state of the development:
1) With standard Yi 4k+ it is not possible to make RTSP stream with 720p, only 480p is possible. The workaround with modified firmware from this repository does not worked for me. Therefore I do not have any latency numbers for 720p.

2) I've got the latency value in range of 120ms to 200ms with standard Yi 4k+ firmware and 480p RTSP stream out of the box. (thus about +/- 40ms around 160ms average latency) The player was done as an app for last iPad pro. I can not say more, since it was commercial project.

My suggestions:
I) Your latency coming probably from RTMP stream. It looks for me like modified firmware creates RTMP stream and multiplexes it to an RTMP stream. Therefore you lose any advantages RTSP protocol provide for you, since it works after RTMP. It is stupid architecture.

II) I am not sure if you will be able to create native RTMP stream with bitstream directly from YI 4k+ codec, since it's pipeline is trimmed for video recording and not streaming. All you an do with firmware - gain access to RTOS linux. Trough this access you will be probably able to grab H.264 or H.265 bitstream at encoder output but then you need to write this output in RAM, which drastically increases your latency. There is also no support from chipmaker of YI 4k+, thus this hardware is no go anymore.

III) Resolution of RTSp stream depends directly on encoder output. Probably if you'll be able to switch input RTSP part  to  higher resolution output of encoder will be the solution. But again (see II) you will stack on memory I/O with high delays for this application.

Good luck guys

gator89, 2018-04-20:
Hi @Nukersson ,

I also would like to have an action cam-like solution where you can stream with low level latency in a better resolution and bitrate combination that is currently possible with YI 4K. 

I think this max 480p resolution is a bottleneck for many people. 

Can I ask which action cam did you choose over YI 4K? Do you have any recommendiation on better streaming quality ones? 

Cheers,
gator

luismanolo, 2019-04-10:
Many thanks for your contribution

I would like to know if the scripts you have created are valid for the "xiaomi action camera 4k". Especially to change the streaming resolution.

Thank you

closed 2019-05-18 by Nukersson