irungentoo/Xiaomi_Yi_4k_Camera #36: Issue StartUp Commands on Yi 4K

somacoder, 2017-11-20:
On the previous Yi action camera, you could enable startup scripts by placing `lu_util exec "command"` in autoexec.ash . However, this does not appear to work anymore. Is there any alternate way to initialize a command on startup (can't alter file system files as the filesystem is read only).

Thanks!

irungentoo, 2017-11-22:
You are talking about commands for the linux?

```t ipc rpc clnt exec1 'command'```

try that.

sukkamehu, 2017-11-22:
Is there still any way to start wifi as client mode on startup? I tried quite long time with many ways but don't get it work.

somacoder, 2017-11-22:
Client Mode is simply triggered by having an stea.conf file on the root of the sd. The menu option doesn’t work. Just leave it in ap mode and add sta.conf. As for starting it automatically, it’s more reliable to dump the options you want to system.pref.download (includes WiFi being on), select the sunset you want to keep, and then place them in system.pref.upload and leave them there. I was looking for a startup command option to turn on FTP automatically, but am finding it just as easy to have a expect script communicate a command via telnet from the access point computer.

sukkamehu, 2017-11-23:
```
##### Wifi configuration file ########################################## 
# SSID 
SSID=Huahei1
# Password 
PASSWORD=12345678
# Device Name 
DEVICE_NAME=AAAAA
# Country Code 
COUNTRY_CODE=BG
```
I made sta.conf as above but it doesn't work, is there something wrong?

Thank you
Juuso

somacoder, 2017-11-23:
The sta.conf looks correct. Do you have the latest firmware? If it’s connected to an Access Point, you should see the wifi icon flashing.

sukkamehu, 2017-11-24:
I have tried with 1.3.22 and 1.3.23 (which has been relased today) but no success. Can you sen your content of system.pref.upload

Below is my result of system.pref.download
https://pastebin.com/QpjnQn8j

somacoder, 2017-11-24:
You are talking about the Yi 4K+. I exclusively use the 4K. Sorry I can’t help you.

closed 2017-11-24 by somacoder

sukkamehu, 2017-11-24:
```
#!/bin/sh
sleep 10
sleep .5
killall -9 hostapd hostapd_autochannel_retartchip dnsmasq udhcpc wpa_supplicant wpa_cli wpa_event.sh 2> /dev/null
killall -9 hostapd hostapd_autochannel_retartchip dnsmasq udhcpc wpa_supplicant wpa_cli wpa_event.sh 2> /dev/null
rmmod bcmdhd
insmod /lib/modules/bcmdhd.ko firmware_path=/usr/local/bcmdhd/fw_apsta.bin nvram_path=/usr/local/bcmdhd/nvram.txt iface_name=wlan dhd_msg_level=0x00 op_mode=1 amba_initmac=04:E6:76:01:01:01
sleep 10
wpa_supplicant -Dnl80211 -iwlan0 -c/tmp/fuse_d/wpa_supplicant.conf -B
sleep 5
udhcpc -i wlan0 -A 2 -b -t 30
sleep 5
killall telnetd
sleep 2
telnetd -F
sleep 5
sh /usr/local/share/network/rtmp.sh "1750269960" "action=restart" "DEV_MAC=58:70:C6:48:74:FE" "RTMP_MAC=" "RTMP_NAME=YIAC3_4748542" "RTMP_SSID=Huahei1" "RTMP_PASSWD=12345678" "RTMP_5G=none" "RTMP_SCAN_EXEC=yes" "RTMP_SCAN_WAIT=10" "RTMP_IP_FIX="192.168.43.43"RTMP_IP_WAIT=20" "RTMP_MMC_WAIT=10" "RTMP_WLAN0_WAIT=10" "WIFI_CHIP=43340" &

```
If some linux master see there's some problem I would like to know. Hopeless to get wifi work on startup.

sukkamehu, 2017-11-27:
@irungentoo should that command `t ipc rpc clnt exec1 'command'` work also for YI4K+? I found out that if I run via telnet this `/usr/sbin/sta.sh TestWlan` it automatically changes to different Wifi called TestWlan but don't get it work in startup.

muzzol, 2023-03-24:
I'm facing same proble. can someone give me instructions about configuring a static IP on 4k?

PaulieCreates, 2024-11-07:
> I'm facing same proble. can someone give me instructions about configuring a static IP on 4k?

Are you looking to add an IP for a live stream server?