irungentoo/Xiaomi_Yi_4k_Camera #37: 1.3.22 beta doesn't work

Negens, 2017-11-22:
Hello,
I don't know if I'm doing something wrong.
I open a script in notepad++ with UNIX LF,
Then I copy the script to a new file and save it as a UNIX script type, with the name autoexec.ash into the root of the sd card.
And nothing happens.
The camera seems to simply ignore the scripts.

irungentoo, 2017-11-22:
Well, it worked when I tried it.

What script are you trying to use?

Negens, 2017-11-22:
I tried the custom shutter values and the bitrate mod.
I even tried to replace manually all the EOL, added the empty line at the end.
Maybe I'm doing something wrong but I don't know what.

To be sure we are on the same firmware, mine is
1.3.22 build 988r

irungentoo, 2017-11-22:
Put the autoexec.ash file in a zip and upload it here (just drag the zip it in your response).

Negens, 2017-11-22:
[autoexec.zip](https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/files/1493562/autoexec.zip)

irungentoo, 2017-11-22:
You need to change the values.

Negens, 2017-11-22:
Well,
so I don't really understand how it works.
I tried to put it with only one line and it doesn't work. So do I have to find the right values in memory ?
How do I do that ? Do I have to dump the firmware or the camera settings to obtain the correct address?

Negens, 2017-11-22:
OK, I read the manuals again and I think I understood

irungentoo, 2017-11-22:
The autoexec.ash file for the bitrate and the shutter hack contain the default values on the camera. You are supposed to change them for the value you want. You can read how to do that in the readme.

Negens, 2017-11-22:
That's OK,
I understood it. Everythings works fine ! 
Thanks a lot for your time !

closed 2017-11-22 by Negens