irungentoo/Xiaomi_Yi_4k_Camera #41: Moving/Renaming files on /tmp/fuse_d over telnet

somacoder, 2017-12-14:
I am attempting to rename files over telnet (Yi 4K) in the /tmp/fuse_d folder (sd card) via the mv command, but it doesn't seem to be possible at all. The entire prompt just hangs, and the system appears to lock up. Does anybody have experience with bypassing this limitation? Is it even possible?

Thanks!

sukkamehu, 2017-12-15:
I tried to execute Linux commands long time in YI4K+ but gave up and got better camera - never again Chinese trash with bad official support.

Why don't they just use all great potential of this camera? It could be so good camera if it would be made to use some native GUI on Linux instead of that shit with all of its limits.

somacoder, 2017-12-15:
Okay, figured it out. This feature is broken in later firmwares.

closed 2017-12-15 by somacoder

MidoriYakumo, 2018-02-28:
filesystem command is only available in AmbaShell(i.e. autoexec.ash)
to invoke these actions via telnet it requires some rpc tricks, I did some test about it but it is quite long time ago

somacoder, 2018-03-29:
Midori, could you share how you were able to access the AmbaShell over telnet? Thanks!