irungentoo/Xiaomi_Yi_4k_Camera #45: Yi 4K new firmware 1.10.4 out

nannini, 2018-01-18:
Hi irungentoo,
Thanks a lot for your awesome work regarding the Xiaomi cameras. I've noticed that a new firmware version 1.10.4 is out for the 4K. Unfortunately the bitrate mod is not working anymore. Will it be supported by your mods?
Thanks a lot

irungentoo, 2018-01-19:
it's done, let me know if something doesn't work.

Eyshika, 2018-04-13:
in this firmware can we add more memory i.e. before it was maximum 64 gb can we go for 128 or 256 gbs now ?

Eyshika, 2018-04-13:
I would like to bring this here, the videos yi4k records get split in 7:00 or less than 7:00 mins automatically. I tried for 30 min continuous video but it got splitted into parts. Couldn't understand the reason behind it