irungentoo/Xiaomi_Yi_4k_Camera #49: Audio bitrate adress in fw 1.4.9

zeke14zeke, 2018-01-27:
Audio bitrate adress is not valid. Bitrate is not changing

irungentoo, 2018-01-27:
on you using PAL or NTSC?

zeke14zeke, 2018-01-27:
I use PAL. I didn't test it in NTSC

irungentoo, 2018-01-28:
I only tested the audio bitrate mod on NTSC so it might be that. I'll see if I can find the audio address for PAL when I have time.

zeke14zeke, 2018-01-28:
Audio bitrate isn't changing in NTSC mode too. Video bitrate adresses are valid: bitrate is changing in PAL and NTSC modes

We-Are-DEVO, 2018-02-08:
Just to confirm, the audio bitrate does not change for me either in either PAL or NTSC. 4K+ 1.4.9 I've only had my camera on 1.4.9, but I'll drop back down to earlier firmwares, if possible, and try switching bitrate. I'll post back and let you know how I get on if go backward with firmware.

We-Are-DEVO, 2018-02-08:
Just flashed my 4K+ to 1.4.8 & 1.3.29 and tested both by directly downloading their .ash files to the SD card , but neither seemed to change the audio bitrate. In both cases the audio bitrate was still showing 48Kb/s.

Earlier on and just in case my NLE was not reporting the correct bitrate, I used the 8Kb/s setting for 1.4.9 as this would definitely be heard even if the NLE didn't report the file info properly, but no luck with that either.

In all instances the camera paused for a few seconds with a red light after power up, which I assume is the camera reading the file.

irungentoo, if you ever need any specific audio testing PM me, I do a fair bit of recording and wouldn't mind testing, especially if it means getting something better than the shitty low 48Kb/s that camera record in.

Here's a couple of my audio tests on YouTube.
[https://www.youtube.com/watch?v=psL1YMzCMgo](url)
[https://www.youtube.com/watch?v=AM4D7rqA2_o](url)