irungentoo/Xiaomi_Yi_4k_Camera #53: Cannot connect to Wi-Fi modem - Yi 4k+

Triuman, 2018-02-09:
Hi, Thank you very much for your effort.

I cannot make my 4k+ connect to wifi modem using WiFi Debug. Can you please explain the properties in the wifi settings? Like sta_freq and chip_type. I am probably missing something.

Thank you.

irungentoo, 2018-02-09:
Which firmware are you using?

Triuman, 2018-02-10:
I am using 1.4.9 now. Tried with WiFi modem and iPhone's hotspot. The wifi icon in the camera flashes without stopping when I turn its wifi on.

I also tried live with custom URL but got "Live connection error". Here is the YiLiver Log:
[https://gist.github.com/Triuman/b09a1a3b51527e20a0311c8365b904f2](url)

irungentoo, 2018-02-10:
CHIP_TYPE is the type of chip, you should not touch that number. STA_FREQ looks like it's to set if the station is 5Ghz or not. Try setting it to 0.

Triuman, 2018-02-11:
Yeah setting it to 0 worked! Awesome, thank you very much!