irungentoo/Xiaomi_Yi_4k_Camera #54: Restoring code on Motherboard of YI 4k+ or resetting Camera

kenoshaYI, 2018-02-10:
My YI 4k+ will no longer start. Screen does slightly light up but no icons. Sometimes I get three flashing blue light on power button. Sometimes led on front will shine. Is there a way to reset camera through jumpers or is there a way to reflash loaded bootup program on motherboard. Thanks KenoshaYI

irungentoo, 2018-02-11:
If you put a firmware.bin on the sd card does it do anything?

kenoshaYI, 2018-02-12:
No it does not. It seems like the camera turns on and I get the front led to turn on but the camera does not start. KenoshaYI

irungentoo, 2018-02-12:
Did you try contacting the yi people to see if they can help you?

kenoshaYI, 2018-02-12:
That I have tried and have not been very successful.

romualdik2, 2018-09-03:
Hi. I have the same problem. How have you solved it?