irungentoo/Xiaomi_Yi_4k_Camera #56: Live RTSP Stream Latency - Custom URL

Triuman, 2018-02-11:
With local live streaming I need a latency below 200 ms. When I stream with VLC, the latency is around 1-2 seconds. I am not sure whether this stems from VLC player's buffer or camera's buffer. Low latency is so crucial for my project. Is there a way to reduce latency?

Thank you.

irungentoo, 2018-02-12:
Are you talking about the rtsp stream on the camera or rtmp streaming with the qr code?

Triuman, 2018-02-12:
The one with QR code but I access it with this URL: "rtsp://192.168.1.26/live"

If I do not use QR code, I cannot see any stream at this address.

Triuman, 2018-02-12:
I got it! I could get latency < 250 ms using Gstreamer. Anyone interested can check the link below.

[https://stackoverflow.com/questions/44160118/gstreamer-pipeline-to-show-an-rtsp-stream](url)

Just run this code on your PC:
gst-launch-1.0 -v playbin uri=rtsp://localhost:8554/test uridecodebin0::source::latency=0

I get least latency with 1080p and highest with 480p. I believe it spends time to compress video size. Even at 1080p though, I get fairly bad video quality. I hope resolve this issue too.

SarHawk, 2018-02-17:
@Triuman 
What parameters did you put on QR code?

Triuman, 2018-02-23:
Really sorry for the late reply.

I used Xiaomi iOS app to create the QR code. Stream quality got better when I set bitrate to 1.3 mbps. Latency was lower at 1080p.

blendfx, 2018-03-06:
@Triuman 
Can you please give me a small tutorial on "how to get the rtsp-stream" running? Is it only possible with gstreamer? I want to get the stream open in VLC or OBC. With rtmp it works well with a nginx rtmp server, but the latency is more than 3 sec. I use the iOS app for the QR Code with the rtmp URL (eg. rtmp://192.168.178.120/live), but with an rtsp URL like "rtsp://192.168.178.120/live" the Live-Stream won't start.

I hope u can help me! Thanks in advance!

Triuman, 2018-03-06:
I really do not have much knowledge about this. I just tried streaming with VLC and Gstreamer. I got a high latency with VLC. I believe you can also use ffmpeg to get the stream like in Gstreamer. In Gstreamer case, I put the ip address of the destination pc in the URL. So, if you are running Gstreamer at a PC with IP 192.168.1.20, you set the URL as "rtsp://192.168.1.20:8899/live". And you use this command in the pc: "gst-launch-1.0 -v playbin uri=rtsp://localhost:8899/live uridecodebin0::source::latency=0"

Hope it helps.

blendfx, 2018-03-07:
Thank you for your reply. I got it :)
Just connect the YI 4k+ with my Laptop in AP Mode and stream via VLC with the url "rtsp://192.168.42.1/live". I just forgot to hit the record button, otherwise it won't stream the live video.

blendfx, 2018-03-14:
Another hint to get the stream working without the need to hit record is to connect parallel with the app... when the app is connected the rtsp stream starts :)