irungentoo/Xiaomi_Yi_4k_Camera #62: question about xiaomi yi 4k scripts

Egorkansk, 2018-03-16:
Good day, I want to ask you a question about changing the bitrate and resolution of the Xiaomi yi 4k action camera, i was read your  amazing scripts, but i not understand how to change the resolution of this camera, for example i want change resolution 720 and set fps 60 (if it possible)

and what that mean: 0xA0659F76 HDR 1920x1080 30P 16:9 in bitrate table , is this possible to set please tell me, thanks

irungentoo, 2018-03-16:
You can set these resolutions with the ```t app test chg_res 0``` command. You replace 0 by the number of the mode (first mode in the bitrate mod file is 0, second mode is 1, etc...)

Most of them don't work though.

Egorkansk, 2018-03-16:
What resolutions, supports by yi 4k besides standart list?