irungentoo/Xiaomi_Yi_4k_Camera #63: Changing shutter speed for video?

x4080, 2018-03-28:
Hi, is it possible to change shutter speed for video ? Use case : in low light situation, we use 24p then the shutter speed standard is 1/48, if we can change it to 1/24 it will be 1 stop brighter (with little more blurness but I think its better than more noise)

Can we change it ? BTW I appreciate your work very much @irungentoo, thanks for everything you did

irungentoo, 2018-04-01:
For the 4k+ there's:
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/tree/master/4k+/shutter_custom_values

The normal 4k doesn't have a shutter speed menu so making a mod isn't as easy.

x4080, 2018-04-02:
Thanks @irungentoo

x4080, 2018-04-03:
I was emailing yi support requesting this feature, if many people email them about this matter, maybe they would grant us