irungentoo/Xiaomi_Yi_4k_Camera #65: One button MOD

mcbiofa, 2018-04-03:
This is the only place where I found someone who's developing something for this action camera! Thank you!
In your opinion is it possible to add the 'One Button' functionality like in GoPro?
For me should be great if the camera should wake up recording the video and when you finish, push the button and shut down.
I think it's the worst thing of this camera....if you use it in real action video (motorbike, parachute, etc etc) it's impossible to know wether you're correctly recording or not.

mcbiofa, 2018-04-03:
I found a (partial) solution in an italian website.
Making a autoexec.ash file with t app key record (for videos) or t app key shutter (for photos) seems working pretty fine for me!

mrcBobLee, 2018-05-11:
@mcbiofa can you please link the mod you found?

mcbiofa, 2018-05-14:
Hi, I made it work making a file with:

##Start custom configuration
t app key record
##End
conf

To make it take pictures instead of video the right command is:
t app key shutter