irungentoo/Xiaomi_Yi_4k_Camera #67: Automatically Start Wifi on Yi4K+

somacoder, 2018-04-06:
Is this possible? If wifi is off, or the power was killed previously, resetting the wifi setting, it will not start automatically. Previously I have used the command 'CustomPref.wifi_auto_start=0x01' in the system.pref.upload file, but that no longer works on the latest firmware Yi 4K+ firmware (1.4.12).

somacoder, 2018-04-06:
Strange. It appears that if you kill power while the wifi icon is flashing, then wifi will not start automatically on the next startup, but if you kill the power outside of that, then it will always autostart. still doesn't solve my problem so happy to hear other possible ways to get around this.

irungentoo, 2018-04-08:
try:
`t wi-fi restart`
in autoexec.ash

somacoder, 2018-04-08:
No, but 
`t wi-fi start`
does work!

Thanks irungentoo, you helped me get really lucky with that one.

somacoder, 2018-04-09:
Hmmm. Now `t wi-fi restart` is no longer working for me. Strange. Still investigating.

somacoder, 2018-04-09:
Alright, phew, figured it out. Couple of things. It appears:
1. You need a line return at the end of the autoexec.ash script to register the command.
2. You need to sleep a little, 10 seconds, perhaps less, for the command to go through successfully.

The wi-fi symbol will not be visible but the network will still be active. I should also note that I am using my own start.sh file on my sd card so that may be influencing this in some way, though I don't think so.

closed 2018-04-09 by somacoder