irungentoo/Xiaomi_Yi_4k_Camera #68: Anybody tried 720/240fps with 135mbps bitrate (yi4k)

x4080, 2018-04-09:
It was hang after a few seconds, using samsung evo+ 64gb, it works until 120mbps though, I just wondering if its the sd card problem or the processor limitation

Thanks

somacoder, 2018-04-09:
That's a pretty insane bitrate. I have those same cards and depending on the firmware on the Yi 4K, I get hanging at much less than that (usually older firmwares). From my experience, it's often a processor issue, especially when there's a lot of movement involved.

x4080, 2018-04-09:
Yes, I tried it to see if image quality will be better. I guess 120 is the max then

x4080, 2018-04-10:
It seems like all resolution is not working with 135mbps, it will hang after a while