irungentoo/Xiaomi_Yi_4k_Camera #73: Yi Discovery

n0ne0ther, 2018-05-04:
Anyone have hands on these? I'm thinking of getting one and would be cool to see how far we can push it for a $60 cam.

nutsey, 2018-05-19:
https://www.goprawn.com/forum/allwinner-cams/8267-yi-discovery-action-camera

Nokius, 2018-10-13:
it looks like the Yi Discovery has the same communication via json as the other Yi Cameras, but I ha dno luck to get the camera view without the Yi app. I did tcpdump with Android and it looks like the Camera is using port 6666 for the stream and exchange a login and password 
`........admin...........................................................12345............................................................`

anyone else hacked around so fare?