irungentoo/Xiaomi_Yi_4k_Camera #74: auto turn on wi-fi script for YI_4K+

alex69russ, 2018-05-15:
It is possible to turn on wi-fi by script? I have one yi 4k+ with broken screen and can not turn on wi-fi on this camera. "t app key start_wifi" not working

somacoder, 2018-05-27:
You can do this on the Yi 4K through the system_pref.upload file, and on the Yi 4K through the autoexec.ash file. Just be sure to put a sleep line of at least 5 seconds before turning the WiFi through autoexec.ash on the Yi 4K+.

somacoder, 2018-05-30:
One quick addendum to this. If you find 5 seconds isn't enough time for the device to boot up (and thus become locked if this command is issued too soon), increase the sleep to say 10000 (10 secs). This is probably a safer bet.

luckylz2git, 2019-03-14:
This also works for Yi 4K Camera fw1.10.9
`sleep 10000`
`t wi-fi restart`

hubertozz, 2019-04-04:
Hi! I have my 4K+ in housing mode with broken screen and the above solution doesn't work. Probably wifi is also set to 5G. Can You help?