irungentoo/Xiaomi_Yi_4k_Camera #81: Can anyone help me enable 60fps in the native 2.7k / 4:3 video mode on the Yi 4k?

LurkerLew, 2018-07-23:
I have a very light grasp of what you guys have accomplished with the Yi 4k using scripts and whatnot, but I am wondering if anyone could tell me if this is possible and perhaps how to do it? Thanks in advance.

jauling, 2018-11-06:
Can the SoC even support this? I almost always record in 1920x1440 60fps, but being able to go higher resolution sounds really exciting to me!

bjcryss, 2019-08-20:
i wonder how come more people don't request this, because for example when filming downhill biking, i need the 4:3 because otherwise in 16:9 the whole bike trail in front of me does not fit in the screen together with also a bit of the handlebar, especially if i turn on the image stabilization on. so i am stuck at filming 30fps... :(   GoPro Hero5 doesn't support 2.7k 4:3 60fps either, but the Hero6 does, i guess they realized this is needed

psolyca, 2019-08-20:
Hey, go there http://psolyca.ovh/yi4k/yi.html and choose bitrate mod. Have fun

bjcryss, 2019-08-20:
> Hey, go there http://psolyca.ovh/yi4k/yi.html and choose bitrate mod. Have fun

Thank you! Very nice tool, good job for who made it!
But even there, i can't find any 2.7k(2704x2032, not even 2560x1920) 4:3 @60fps setting

psolyca, 2019-08-20:
Hahaha yes sorry. I am on holiday so I have my head elsewhere. It is on my Todo list to be able to change fps but I did not find it yet.

bjcryss, 2019-08-21:
ah, so you created the website :D nice, it really helps, thanks!
i was thinking that it's not actually possible to have 4:3 @60fps, but i'm eager to see your updates