irungentoo/Xiaomi_Yi_4k_Camera #83: How to livestream at higher than 1.2Mbps over RTMP?

reedog117, 2018-09-03:
I noticed that people have figured out the QR code system for initiating a livestream. However, has anyone figured out how to stream at higher bitrates? Ideally I'd like to stream 720p60 at 5500-6000kbps. However I've seen that the only documented options are 

```
{
    "ssid":"YOUR_WIFI_NETWORK NAME",
    "pwd":"YOUR_WIFI_NETWORK_PASSWORD",
    "res":"1080p",
    "rate":"3",
    "dur":"0",
    "url":"rtmp://some-rtmp-link"	
}
```
where "rate” takes 0 for auto, 1 for low (0.6 Mbps) rate, 2 for middle (0.8 Mbps) rate, 3 for high (1.2 Mbps) rate.

Does anyone know if auto is adaptive and goes way higher than these other options? Or what happens if I start putting in numbers greater than 3? My 4K camera is coming in later this week, so I'm hoping someone already knows the answers to this.

reedog117, 2018-09-07:
Well.. I half-figured it out. Unpack the firmware (for the 4K at least) and check out the linux partition (out_8.bin for 4K)

There's a file in /usr/local/share/script called rtmp.connect.sh -- this section looks promising

```
RTSP_CONF()
{
    if [ "${RTMP_CONFIG_SIZE}" == "720p" ]; then
        /usr/bin/SendToRTOS rtmp 80
    elif [ "${RTMP_CONFIG_SIZE}" == "1080p" ]; then
        /usr/bin/SendToRTOS rtmp 81
    elif [ "${RTMP_CONFIG_SIZE}" == "360p" ]; then
        /usr/bin/SendToRTOS rtmp 83
    else
        /usr/bin/SendToRTOS rtmp 82
    fi

    if [ "${RTMP_CONFIG_RATE}" -eq 3 ]; then
        /usr/bin/SendToRTOS rtmp 96
    elif [ "${RTMP_CONFIG_RATE}" -eq 2 ]; then
        /usr/bin/SendToRTOS rtmp 97
    elif [ "${RTMP_CONFIG_RATE}" -eq 1 ]; then
        /usr/bin/SendToRTOS rtmp 98
    else
        /usr/bin/SendToRTOS rtmp 99
    fi
}
```

So now the next question is how can I see where this corresponds in the RTOS and how can I add additional values?

psolyca, 2018-09-08:
I guess you could follow the instructions given for audio and video bitrate
https://github.com/irungentoo/Xiaomi_Yi_4k_Camera/blob/master/bitrate/Readme.md
and
https://github.com/wavepacket0/Xiaomi_Yi_4k_Plus/blob/master/bitrate_mod/how_to_find_audio_bitrate_address.txt
and
https://github.com/wavepacket0/Xiaomi_Yi_4k_Plus/blob/master/bitrate_mod/how_to_find_video_bitrate_address.txt
I do not remember but I have seen a tutorial on using IDA pro to decompile the RTOS here, on DashCamTalk or wavepacket0 repository...

reedog117, 2018-09-10:
While the bitrate info is helpful, the problem is that I don't know what memory addresses to write to for the livestreaming bitrates. I've tried doing a hexedit search on out_4.bin, but I think that the firmware is compressed as I'm not finding any useful strings. I can try pulling out IDA pro but I think that will be a steep learning curve...

lainsilk, 2018-11-01:
Hi Guys, has anyone has figureout if the streaming connections accepts proxy settings?

psolyca, 2018-11-06:
Hey,

I figured it out ^^

So the calculation is different from the main video bitrate.
In RTMP case, there is a base value of 1.0 Mbps
For each resolution, a table close to the one for the main video but the first value is not the highest bitrate but the higher multiplicator.

The address 0xA011B8E0 correspond to the base bitrate value, 0x49742400 (1 000 000 bps).
The address 0xA06E50BC correspond to the address for the multiplicator for the high quality settings.
To get the bitrate address for the medium quality option, add: 0x10 i.e. 0xA06E50CC.
To get the address for the low quality option, add: 0x20 i.e. 0xA06E50DC.

Other values that can be set other than the bitrate are the lower and upper bound values.
These values correspond to how closely the encoder should try to follow the set bitrate.
To get the address of the lower bound, add 0x04 to the bitrate address, i.e. 0xA06E50C0.
To get the address for the upper bound add 0x08 to the bitrate address, i.e. 0xA06E50C4.

Tables for addresses and bitrate are on my gist :

- [Addresses](https://gist.github.com/psolyca/feb44a9923dfb461febfa95128125ec8)
- [Base bitrate](https://gist.github.com/psolyca/722b9b9e32e6c66e29f9b4c5788ea50f)
- [Bitrate multiplicator](https://gist.github.com/psolyca/57a4025b0bba5216db728e1054c04a2d)

brianspit, 2018-11-06:
@psolyca do you have an example of what the file should look like?

I tried something like this:
```
#1280x720 30P 6 Mbps 
writel 0xA06E50BC 0x41F00000 #<- is this the correct way to use the resolution?
writel 0xA011B8E0 0x49927C00 #1.20 Mbps- base bitrate value 
writel 0xA06E50BC 0x40A00000 #5 - multiplicator high quality settings
writel 0xA06E50C0 0x3F800000 #lower bound: 1
writel 0xA06E50C4 0x3F800000 #upper bound: 1
```
But it didn't work.

psolyca, 2018-11-06:
@brianspit No that is not the good way.

To change the 1280x720 30P to 6 Mbps, you should write :
`
writel 0xA011B8E0 0x49927C00 #<- this is the base of the bitrate
writel 0xA06E50BC 0x40A00000 #5 - multiplicator high quality settings
`
That is all.
Cause lower and upper bound are always egal to 1.0
So a complet matrix for all resolution could be this one :
`
writel 0xA011B8E0 0x49927C00 #base of the bitrate
writel 0xA06E50BC 0x40A00000 #5 - multiplicator high quality settings for 1280x720p
writel 0xA06E50CC 0x40933333 #4.6 - multiplicator middle quality settings for 1280x720p
writel 0xA06E50DC 0x40866666 #4.2 - multiplicator low quality settings for 1280x720p
writel 0xA06E50FC 0x40A00000 #5 - multiplicator high quality settings for 1920x1080p
writel 0xA06E510C 0x40933333 #4.6 - multiplicator middle quality settings for 1920x1080p
writel 0xA06E511C 0x40866666 #4.2 - multiplicator low quality settings for 1920x1080p
writel 0xA06E513C  0x40A00000 #5 - multiplicator high quality settings for 640x360p
writel 0xA06E514C  0x40933333 #4.6 - multiplicator middle quality settings for 640x360p
writel 0xA06E515C  0x40866666 #4.2 - multiplicator low quality settings for 640x360p
writel 0xA06E517C  0x40A00000 #5 - multiplicator high quality settings for 720x480p
writel 0xA06E518C  0x40933333 #4.6 - multiplicator middle quality settings for 720x480p
writel 0xA06E519C  0x40866666 #4.2 - multiplicator low quality settings for 720x480p
`

Original matrix is :
`
Address           H   LH UH    M   LM UM   L    LL   UL   Res
0xA06E50BC   1.6 1.0 1.0    1.2 1.0 1.0     0.8 1.0 1.0   1280x720p
0xA06E50FC   1.2 1.0 1.0    0.8 1.0 1.0     0.6 1.0 1.0   1920x1080p
0xA06E513C   0.5 1.0 1.0    0.3 1.0 1.0     0.2 1.0 1.0   640x360p
0xA06E517C   1.0 1.0 1.0    0.8 1.0 1.0     0.5 1.0 1.0   720x480p
`

brianspit, 2018-11-06:
@psolyca thank you!

psolyca, 2018-11-08:
Ok thanks for the feedback.
I will investigate why but maybe a hardware limitation on the Yi !
I did not pay attention on the resolution and bitrate when you are using the phone app and changing the parameters on the phone !
Does the stream is downsizing for transmission purpose as the video is still recorded on the phone ?
I have seen a couple of coefficient in others functions but I did not investigated them for now.

psolyca, 2018-11-09:
@brianspit, do you have stable bitrate even at 5 Mbps ?
Cause someone reported that his bitrate is [not stable at 720p](https://dashcamtalk.com/forum/threads/autoexec-ash-commands-collection-update-18-11-06-rtmp-bitrate.36825/post-439655).

brianspit, 2018-11-09:
@psolyca no it's not stable 5 Mbps it switches from around 4 to 6 at 720p. 
btw I'm working together with dayz_hk he has been testing everything.

alex-rptrv, 2018-11-09:
Hi, Where do I update the code to gear up to 4-5 mbits per second ? 

Skickat från min iPhone

psolyca, 2018-11-09:
Hey,

Sorry. https://r.tapatalk.com/shareLink?url=https://dashcamtalk.com/forum/threads/autoexec-ash-commands-collection-update-18-11-06-rtmp-bitrate.36825/&share_tid=36825&share_fid=65060&share_type=t

brianspit, 2018-11-09:
@psolyca would you like to join this discord server https://discord.gg/Kdt7MU we like to get everything working properly and test a lot too.

psolyca, 2018-11-11:
Resolution order is not the right one :

- 0xA06E50BC for 1080
- 0xA06E50FC for 720
- 0xA06E513C for 480
- 0xA06E517C for 360

And resolution for 480p is more 848x480p than 720x480p
So the matrix could be this one :
```
writel 0xA011B8E0  0x49927C00 #base of the bitrate
writel 0xA06E50BC  0x40A00000 #5 - multiplicator high quality settings for 1920x1080p
writel 0xA06E50CC  0x40933333 #4.6 - multiplicator middle quality settings for 1920x1080p
writel 0xA06E50DC  0x40866666 #4.2 - multiplicator low quality settings for 1920x1080p
writel 0xA06E50FC  0x40A00000 #5 - multiplicator high quality settings for 1280x720p
writel 0xA06E510C  0x40933333 #4.6 - multiplicator middle quality settings for 1280x720p
writel 0xA06E511C  0x40866666 #4.2 - multiplicator low quality settings for 1280x720p
writel 0xA06E513C  0x40A00000 #5 - multiplicator high quality settings for 848x480p
writel 0xA06E514C  0x40933333 #4.6 - multiplicator middle quality settings for 848x480p
writel 0xA06E515C  0x40866666 #4.2 - multiplicator low quality settings for 848x480p
writel 0xA06E517C  0x40A00000 #5 - multiplicator high quality settings for 640x360p
writel 0xA06E518C  0x40933333 #4.6 - multiplicator middle quality settings for 640x360p
writel 0xA06E519C  0x40866666 #4.2 - multiplicator low quality settings for 640x360p
```

psolyca, 2018-12-06:
After tries and error, I finally manage to change bitrate in RTMP mode.
I made a webpage to create an autoexec.ash file http://psolyca.ovh/yi4k/yi.html
This page will evolved later...

Eater4, 2019-04-28:
So does anyone just have a autoexc file that changes the bit rate to 5mpbs?

psolyca, 2019-04-29:
Build it yourself ;-)
https://github.com/psolyca/Yi_4k_ROOTFS/wiki/Hacking-the-memory

rotahn, 2019-05-06:
Hi guys

Firstly,  thanks for all the work and time you have put into this!

Does this work for the 4k plus? If so, does it work for firmware 1.4.14 build 1019r?

Thanks

psolyca, 2019-05-06:
Sorry, the custom firmware does not work with 4k plus.

rotahn, 2019-05-06:
I thought it was just a text file that was placed on the root of the sdcard?

I was just wondering if the memory addresses you guys have found here would be the same.. i guess they won't be though..ahh well.. i will carry on searching lol

psolyca, 2019-05-06:
No memory addresses are different from a device to an other and from a firmware to an other (not always).
Thus, values irungentoo and I (and others) have found will not work with another firmware.

Sorry, you have to disasm the Yi4k+ firmware to find those values (irungentoo have some values). Someone has began to look at it but I do not have any information if he stops or not.

rotahn, 2019-05-08:
Well..i have unpacked the firmware and currently have it loaded in the freeware version of IDA PRO and while i know what im looking at is assembler / hex, i have no idea. I cant afford IDA PRO otherwise i wouldn't mind spending the time to figure it out myself.
Is there any really kind person around who wouldn't mind finding out the addresses 😋

Bl1nk3y, 2020-03-21:
Did anyone manage to find the values? 
I have been unable to so far.

copy, 2021-04-15:
I disassembled the rtos using ghidra (good alternative to IDA) and found some addresses to increase the bitrate for Yi4k+. In firmware v1.4.14 (the only one you can download from [here](https://www2.yitechnology.com/support/firmware_action/id/2.html)) there's a function at 000e8304 that parses the `res` and `rate` keys from the QR json. The `res` parsing code looks like this:

```c
            iVar1 = FUN_0058e6f0_strcmp(iVar4,"1");
            if (iVar1 == 0) {
              local_3c_bitrate = 800000;
            }
            else {
              iVar1 = FUN_0058e6f0_strcmp(iVar4,"3");
              if (iVar1 == 0) {
                local_3c_bitrate = 2000000;
              }
              else {
                local_3c_bitrate = 1200000;
              }
            }
```

0.8, 1.2 and 2.0 are the mbit/s values for "1", "2" and "3" respectively. They are hard-coded as integers in the code. The upper 16 bits of `local_3c_bitrate = 2000000` are set in a `movt` instruction at 000e8588. By default it contains 0x1e (2000000 >> 16). I changed it to `0x4c` (5000000 >> 16) to get a bitrate of approximately 5 mbit/s.

tldr:

For firmware v1.4.14 with `"res":"3"` you can increase the bitrate to 5 mbit/s like this:

```
writeb 0xE8588 0x4B
```

In firmware v1.4.9 I found the same instruction at 0xE8528. For other firmwares you can search for `1e 30 40 03` using a hex editor to obtain the address.

xnitro1, 2024-07-31:
> I disassembled the rtos using ghidra (good alternative to IDA) and found some addresses to increase the bitrate for Yi4k+. In firmware v1.4.14 (the only one you can download from [...]

What exactly do you mean with "res:3" ? 

I tried the following QR Coder: https://sanjayravindran.github.io/Yi-4k_Plus_QRcodeGenerator/

This gave me a code like this: {"ssid":"WIFIHERE","pwd":"PWHERE","res":"1080p","rate":"3","dur":"0","url":"rtmp://a.rtmp.youtube.com/live2/XXXXXXXXXX","ak":"0","sign":""}

the "res" stated in the code is 1080p. i also tried to add your writeb to my autoexec, but it just doesnt change anything. 

my autoexec looks like this:

#Bitrate mod by irungentoo for the yi4k+ firmware 1.4.14
#Example autoexec.ash with default values
#
#
#1920x1080 48P 16:9
#high quality
writel 0xABA7F4 0x42700000 #60mbps
writel 0xABA7F8 0x41200000 #lower bound: 0.75
#medium quality
writel 0xABA804 0x42340000 #45mbps
writel 0xABA808 0x40A00000 #lower bound: 0.75
#low quality
writel 0xABA814 0x41C00000 #24mbps
writel 0xABA818 0x40400000 #lower bound: 0.75
writeb 0xE8588 0x4B
#
#Audio:
writel 0xA972B0 0x1F400 #128000bs

am i doing anything wrong?

i just want to stream with a higher bitrate :(