irungentoo/Xiaomi_Yi_4k_Camera #102: How to increase volume of streaming to YouTube on Xiaomi Yi 4k ?

kiber-chaka, 2020-04-25:
Hi!
I stream to YouTube thru generated picture with parameters:
`{"ssid":"wifi-home","pwd":"wifi-pass","url":"rtmp://a.rtmp.youtube.com/live2/key","res":"1080p","rate":"3","dur":"0"}`
But volume is a lot highter on recorded video, so I suppose theoretically it is possible to increase volume for streaming. 
I tryed to look in volume settings during the stream and volume there was set to highest mod.
So, may be I should to add some parameter into QR code?
Thank you!

apsheronets, 2020-06-05:
RTMP mod is what you need: http://psolyca.ovh/yi4k/yi.html
By the way, it doesn't work on 4k+.

TijuanaKez, 2020-06-05:
Hi thanks for the reply!

Yes I’ve investigated RTMP mod and thats great. But my touch screen is cracked so I use ‘In Housing Mode’
How can I create an autoexec.ash that will force the camera into streaming mode on boot?

Thanks,

Kez.

apsheronets, 2020-06-05:
If only the wifi was enabled before your accident with the camera screen. Use the android application in this case.

TijuanaKez, 2020-06-05:
Hi thanks again,

I remember I once had an used and auto exec.ash to dump the current settings to the SD card, then I could modify the settings and the camera would always start per those settings when it was turned on.

Do you know if this can be done to set the camera into ‘Live’ mode?

The mobile app can only produce the QR code, it cannot set the camera into ‘Live’ mode to look for the QR.

Thanks